cpaxton/costar_stack #10: ROS Kinetic support

cpaxton, 2017-02-08:


ahundt, 2017-02-18:
@bcjiang what's the status of this?

bcjiang, 2017-02-18:
@ahundt There's still a build issue with robotiq/soem. Some perception packages (such as sp_segmenter, ar_track_alvar, predicator_collision)  are also not yet successfully built. Other packages work fine.

ahundt, 2017-02-19:
cool thanks for the info... about how much work does it seem like it will take to fix?

bcjiang, 2017-02-19:
It seems installing robotiq/soem on kinetic is a tough issue.. some people said it could be fixed by mannual modification of the packages. I haven't yet investigated the perception package issues, since dvrk branch doesn't need them for now and I just catkin ignore them.

cpaxton, 2017-02-19:
Kinetic isn't supported for ROS industrial or by robotiq. Also Baichuan the DVRK packages don't really need robotiq either-- we can do some refactoring.

In general it looks like the industrial robot version  of CoSTAR is going to be targeted at Ubuntu 14.04 for the foreseeable future though. Unless you think you  can get SOEM to work Andrew?

Best,
Chris

From: Baichuan Jiang
Sent: Saturday, February 18, 10:45 PM
Subject: Re: [cpaxton/costar_stack] ROS Kinetic support (#10)
To: cpaxton/costar_stack
Cc: Chris Paxton, Author

It seems installing robotiq/soem on kinetic is a tough issue.. some people said it could be fixed by mannual modification of the packages. I haven't yet investigated the perception package issues, since dvrk branch doesn't need them for now and I just catkin ignore them.

fjonath1, 2017-03-02:
@bcjiang
Regarding sp_segmenter, I recently updated it and had it compiled on Kinetic.
Remember to install ObjRecRANSAC library and if you're building OpenCV2.7 by yourself (not using the prebuilt one), you should enable opencv contrib when you're building it.

bcjiang, 2017-03-02:
@fjonath1 Thanks, tried to install ObjRecRANSAC from the repository[ here](https://github.com/tum-mvp/ObjRecRANSAC), but catkin build report error: ‘class vtkPolyData’ has no member named ‘Update’. Is there any required version for VTK? I installed the latest version of VTK and also have the opencv/cmake ready to use.

fjonath1, 2017-03-03:
@bcjiang I also have a relatively recent vtk (vtk 7.0.0) although I runs it in OSX.

Can you post the error report?
Also, is the pcl you have also compiled with this vtk version? Since pcl also uses vtk, so if you changed vtk, it tends to broke the pcl, which may contribute to this problem.

If you have some time, maybe we can schedule a meeting to try resolving this problem. Also, Jin had successfully installed sp_segmenter in his laptop with my help, and I believe he uses Kinetic too.

bcjiang, 2017-03-04:
@fjonath1 I installed vtk 7.1.0, and the error message when I'm building ObjRecRANSAC looks like this: 
/home/ziri/catkin_ws/src/ObjRecRANSAC/src/VtkBasics/VtkLine.cpp:27:9: error: ‘class vtkTubeFilter’ has no member named ‘SetInput’
  mTube->SetInput(mLine->GetOutput());
/home/ziri/catkin_ws/src/ObjRecRANSAC/./src/VtkBasics/VtkPoints.h:36:40: error: ‘class vtkPolyData’ has no member named ‘Update’
I don't think I have compiled pcl. Yes a meeting would be great. Do you have time this weekend? I am available anytime.

ahundt, 2017-03-06:
@bcjiang you need to look at the vtkTubeFilter class in the VTK documentation/implementation and see how they changed the API so you can update the function calls to match.

fjonath1, 2017-03-06:
Costar_stack now builds in kinetic with commit [569ec21f191fa1b74da134c6cd8cd90025e9f932](https://github.com/cpaxton/costar_stack/commit/569ec21f191fa1b74da134c6cd8cd90025e9f932)
We disabled alvar_data_collection and simple_handeye_calibration.
We enabled c++11 compilation on costar_predicator_collision, and fix the sp_segmenter problem.

cpaxton, 2017-03-07:
Sounds great. We should add a kinetic install script that disables those like how the default indigo one disables perception.

Best,
Chris

ahundt, 2017-03-10:
I began testing this and it definitely isn't ready yet as there are not working setup instructions, I began testing by simply converting "indigo" to "kinetic" and it fails at several locations. Please reopen this issue.

First, how do I install fcl with kinetic?

ahundt, 2017-03-10:
Perhaps it seems soem should be skipped with kinetic?

ahundt, 2017-03-10:
There is also a [bug in the setup script](https://github.com/cpaxton/costar_stack/blob/master/install_indigo.sh#L21).

`source ../devel/setup.bash`

produces the following error:
```
+ source ../devel/setup.bash
./costar.sh: line 75: ../devel/setup.bash: No such file or directory
```

Where is this file supposed to be? (note I'm running a slightly modified, but roughly equivalent version so the filename and line numbers differ).

ahundt, 2017-03-10:
I tried building the following, I'm not sure what is expected here and what is not:

```
Failed     << ar_track_alvar:make                                       [ Exited with code 2 ]
Failed    <<< ar_track_alvar                                            [ 18.3 seconds ]
Abandoned <<< alvar_data_collection                                     [ Depends on failed job ar_track_alvar ]
Abandoned <<< simple_handeye_calibration                                [ Depends on failed job ar_track_alvar ]
Finished  <<< iiwa_hw                                                   [ 11.0 seconds ]
Finished  <<< object_on_table_segmenter                                 [ 30.1 seconds ]
[build] Summary: 52 of 68 packages succeeded.
[build]   Ignored:   5 packages were skipped or are blacklisted.
[build]   Warnings:  6 packages succeeded with warnings.
[build]   Abandoned: 10 packages were abandoned.
[build]   Failed:    6 packages failed.
[build] Runtime: 33.9 seconds total.
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

fjonath1, 2017-03-10:
I forgot whether fcl and soem could be installed. Maybe @bcjiang know?

I know that ar_track_alvar couldn't build on kinetic yet, however I think our main code is not directly rely on ar_track_alvar. We could always use another ar tracker I guess for kinetic alternative.

For simple_handeye_calibration, I forgot to actually update it. You could easily fix it by removing ar_track_alvar dependency on its package.xml and CMakeLists.txt, since it should totally work with any ar tracker that publishes tf.

ahundt, 2017-03-10:
Here is the exact script I'm using atm:
https://github.com/ahundt/robotics_setup/blob/master/costar.sh

bcjiang, 2017-03-10:
I just built it again and only the soem failed. (Some packages lilke ar_track_alvar are catkin-ignored in the kinetic branch I think). I didn't remember anything wrong with fcl, actually I don't have that in my catkin workspace.

ahundt, 2017-03-10:
So the setup.bash file is not created until after the first time you build. This code has obviously never been tested on a fresh ubuntu system.

fjonath1, 2017-03-10:
Yeah, the setup.bash file is not created until first catkin build. That is a bug.

I read your script. Btw, we have kinetic-devel branch also.
I think there are couple of fixes that's being done over there.

For [fcl](https://github.com/flexible-collision-library/fcl.git), it is now an independent library separate from ros. Do we actually have dependency on it?

@bcjiang Just curious, could you ar_track_alvar kinetic-devel branch to test whether it could build or not on your machine? I think the one you've tried is master branch right? If it could build, you could reenable other stuff that depends on it.

bcjiang, 2017-03-10:
Sure I tried ar_track_alvar again and it failed with the c++ 11 issue.

ahundt, 2017-03-10:
definitely missing quite a lot:

```
[build] Summary: 54 of 67 packages succeeded.
[build]   Ignored:   5 packages were skipped or are blacklisted.
[build]   Warnings:  None.
[build]   Abandoned: 8 packages were abandoned.
[build]   Failed:    5 packages failed.
[build] Runtime: 44.0 seconds total.
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

How do I see a more specific list of what failed?

ahundt, 2017-03-10:
@bcjiang it seems the kinetic-devel branch worked for me.
```
Finished  <<< ar_track_alvar                                            [ 34.7 seconds ]
```
Though ar_track_alvar seems ok, all the missing packages mentioned above still don't work.

bcjiang, 2017-03-10:
Maybe you can scroll up to see the build log?

ahundt, 2017-03-10:
@fjonath1 I assume we do have an fcl dep, right? I believe it is utilized for the collision detection of moveit, which I believe is currently one of the broken packages for me.

ahundt, 2017-03-10:
here is my build log with the components that failed:
https://gist.github.com/ahundt/55580c030b7201d6ac4ab045de6ce730

fjonath1, 2017-03-10:
Yeah, I think some moveit packages depends on fcl. Ros already separate fcl from its repo, so I guess we need to build and install it outside of the ros environment to use it.

Predicator collision and moveit_collision_environment should build as far as I remember. The predicator collision problem is that we need to enable c++11 (I think?) which was fixed in the last 2 commits on our kinetic branch. I'm not sure about moveit_collision_environment, but I believe it successfully built on @bcjiang machine.

ahundt, 2017-03-10:
does moveit come with ros or is there a separate process for that?

How should I solve the predicator collision problem?

I've been using master, as specified in the [costar.sh install script(https://github.com/ahundt/robotics_setup/blob/master/costar.sh) I linked, is that not what I should be doing?

bcjiang, 2017-03-10:
I can build moveit_collision_environment and predicator_collision (seems we solved this by enabling c++11), as for other packages:
- ur_modern_driver looks like a similar c++11 problem (It is catkin ignored somehow)
- robotiq_ethercat is related to soem problem.
- robotiq_s_model_articulated_gazebo_plugins this is also soem issue.

fjonath1, 2017-03-10:
I think moveit is usually installed with apt-get on linux.
On OSX, we need to build it from source.

For predicator and moveit_collision_environment problem, try kinetic_devel branch to see whether it helps.

bcjiang, 2017-03-10:
Oh actually I removed robotiq_s_model_articulated_gazebo_plugins from my workspace when I failed to build it, so I don't know what's exactly the problem with it.

ahundt, 2017-03-10:
@fjonath1 so what explains the errors I'm getting, as you can see in the script I run the apt-get line to install moveit.

ahundt, 2017-03-10:
@bcjiang what commands do I use to fix those items you mentioned? I'd like to add them to the [costar.sh](https://github.com/ahundt/robotics_setup/blob/master/costar.sh) script.

bcjiang, 2017-03-10:
I think @fjonath1 knows how to deal with the c++11 issue, he helped me build those packages.

fjonath1, 2017-03-10:
@ahundt For moveit_collision_environment, it only depends on moveit packages (  moveit_core, moveit_msgs, moveit_ros_planning_interface), ros standard library (tf), and our msgs (costar_objrec_msgs). I don't see why it should be fail to build. Could try to `catkin build moveit_collision_environment` and see what the actual error message is, since the log that you provided didn't tell me anything about the issue.

For the c++11 issue on predicator collision, refer to this commit [569ec21f191fa1b74da134c6cd8cd90025e9f932](https://github.com/cpaxton/costar_stack/commit/569ec21f191fa1b74da134c6cd8cd90025e9f932)

ahundt, 2017-03-10:
Okay I compiled just that package and saved the relevant log, [here is the line that fails in the log output moveit_collision_environment.txt](https://gist.github.com/ahundt/55580c030b7201d6ac4ab045de6ce730#file-moveit_collision_environment-txt-L273).

cpaxton, 2017-03-10:
setup should be ok. ../devel/setup.bash is supposed to be at the end.

ahundt, 2017-03-10:
we might be able to apply this for the ur_modern_driver:
https://github.com/iron-ox/ur_modern_driver/tree/iron-kinetic-devel

ahundt, 2017-03-10:
wait should I be using https://github.com/cpaxton/costar_stack/tree/kinetic-devel?

ahundt, 2017-03-10:
Also has xdot been merged directly into costar_stack?

fjonath1, 2017-03-10:
> Okay I compiled just that package and saved the relevant log, here is the line that fails in the log output moveit_collision_environment.txt.

Interesting, seems like the header for some standard ros classes has changed in kinetic.
Is compiling with c++11 solve the problem in moveit_collision_environment? 

> wait should I be using https://github.com/cpaxton/costar_stack/tree/kinetic-devel?

I have been telling you to try that branch yesterday. It has few fixes, but since your problem now is just moveit_collision_environment, I think you could just add c++11 into its CMakeLists.txt and recompile.

ahundt, 2017-03-10:
Okay I switched to the kinetic-devel branch and got these errors:
https://gist.github.com/ahundt/55580c030b7201d6ac4ab045de6ce730

ahundt, 2017-03-10:
I created this pull request for ur_modern_driver
https://github.com/ThomasTimm/ur_modern_driver/pull/96

ahundt, 2017-03-10:
Looks like some people have made patches which could work for us on kinetic with ethercat & robotiq drivers:
https://github.com/ros-industrial/robotiq/issues/81

ahundt, 2017-03-11:
we made kinetic-devel and indigo-devel selectable here:
https://github.com/jhu-lcsr/robotiq/tree/kinetic-devel

ahundt, 2017-03-11:
Appropriate setup script here:
https://github.com/ahundt/robotics_setup/blob/master/costar.sh

closed 2017-03-22 by cpaxton

ahundt, 2017-03-23:
Merged in:
https://github.com/cpaxton/costar_stack/pull/65