cpaxton/costar_stack #24: Install.md

ahundt, 2017-02-17:
I'm looking through [install.md](https://github.com/cpaxton/costar_stack/blob/master/INSTALL.md) and it seems the setup manually clones a number of repositories. Is there a vcstool or wstool (forgot which) which provides correct working commit information?

ahundt, 2017-02-17:
Also I'd like to add a setup script as part of this issue to https://github.com/ahundt/robotics_setup

ahundt, 2017-02-17:
I'll probably like to configure it to either set up in one of:

- `~/catkin_ws/src`
- `~/src/catkin_ws/src`

cpaxton, 2017-02-17:
That's the standard place which means it may clash with someone else's ros workspace.

Best,
Chris

ahundt, 2017-02-17:
can you clarify... what's the standard practice?

If there is a way to create the setup script such that it is minimally interfering and most likely to work even if someone has an existing setup, I'd appreciate those details.

cpaxton, 2017-02-17:
Yeah, people are likely to already have a catkin workspace, but you can chain or have multiple -- so it's better to make a costar specific one.

ahundt, 2017-02-17:
okay, but can vcstool or whatever the appropriate thing is be used for a costar specific one?

cpaxton, 2017-02-17:
To clarify, catkin workspace is normally:
```
/home/$USER/catkin_ws
```

If theirs has any shared third party dependencies with different versions, there might be a clash, so it's better to build in:
```
/home/$USER/costar_ws
```

wstool doesn't solve this problem because the concern is multiple versions.

cpaxton, 2017-02-17:
yeah we can still use it though

ahundt, 2017-02-18:
okay, is it this file? Are some things missing?
https://github.com/cpaxton/costar_stack/blob/master/costar.rosinstall

We should also have a master version and create at least 1 release where specific release branches (preferably), tags (also good), or hashes (if completely necessary) are used to produce a known working configuration.

cpaxton, 2017-03-02:
@ahundt what is the issue here? can we close this?

ahundt, 2017-03-06:
The issue here is that compatible release versions aren't specified so breakage will eventually occur

ahundt, 2017-04-19:
I believe this item is still outstanding because we don't have a reliable way of installing the exact configuration we use on the lab workstation... do we? For example this would include all custom patches like in ur_modern_driver, and all other repos.

cpaxton, 2017-04-24:
@ahundt I added a note about ur modern driver pointing to the correct branch. The rest of our repos are up to date. I use the install.md file extensively to set up CoSTAR on new machines, so I'm happy with this. I also don't have any intention of specifying release versions for the time being. My `install_indigo.sh` script is also reliable and has been tested. What's the problem?

cpaxton, 2017-04-27:
Closing since no response

closed 2017-04-27 by cpaxton