cpaxton/costar_stack #32: Robot Mode: [NOT CONNECTED]

davidadley, 2017-02-23:
Hello,

I am trying to use costar_stack without a physical robot in a simple simulation. Everything loads up ok but the robot mode is 'not connected'.  I believe this is an error with the `iiwa_driver.py` script in particular as the vitual joints it uses are not available as I detailed [here](https://github.com/SalvoVirga/iiwa_stack/issues/57) but maybe there is a simpler solution?

After entering:
```
roslaunch iiwa_gazebo iiwa_gazebo.launch trajectory:=false
roslaunch costar_bringup iiwa14_s_model.launch sim:=true start_sim:=false
```

I get the following error to stdout:

```
[ INFO] [1487866168.621380739, 47.564000000]: MoveitSimpleControllerManager: Waiting for PositionJointInterface_trajectory_controller/follow_joint_trajectory to come up
[ INFO] [1487866173.634137317, 52.565000000]: MoveitSimpleControllerManager: Waiting for PositionJointInterface_trajectory_controller/follow_joint_trajectory to come up
[ERROR] [1487866178.648899862, 57.565000000]: MoveitSimpleControllerManager: Action client not connected: PositionJointInterface_trajectory_controller/follow_joint_trajectory
[ INFO] [1487866178.710239246, 57.627000000]: Returned 0 controllers in list
[ INFO] [1487866178.726409028, 57.643000000]: Trajectory execution is managing controllers
L
```

However I believe this might be ok as trajectory is set to `false` in the above commands therefore there should not be a trajectory controller. 

Am I missing something simple to change the robot from not connected to connected and then I can servo to different waypoints using the simulator?

cpaxton, 2017-02-23:
I'm noticing this on my own system too, which means we probably broke something with one of our more recent CoSTAR releases. I'll take more of a look at this when I get a chance. We recently changed how these positions are managed, moving them from static transform publishers to the CostarArm class.

cpaxton, 2017-02-23:
The trajectory controller issues should be fine.

cpaxton, 2017-03-02:
@davidadley this should now be fixed in master. there were some new features in the base CostarArm class that were causing some issues.

davidadley, 2017-03-06:
Great, working here now!

closed 2017-03-06 by davidadley