cpaxton/costar_stack #35: SmartMove UI does not update list of waypoints

cpaxton, 2017-03-02:
Because there's no "Refresh waypoints" button and it used to do that on creation of a new UI

cpaxton, 2017-03-06:
Hey Felix, so we want to solve this by exposing a refresh() function from the different UI elements, right?

Where would that be called from? I'd like to get this fixed over the next couple days if we can.

fjonath1, 2017-03-06:
We could make it refresh everytime we click the action button on UI on 
`def component_selected_callback(self,name): `

Alternatively, we can also call the refresh function on wherever instructor view used to remake the plugin with
` self.all_generators[name] = self.plugins[name]['module']() `

Although I'm not sure whether we actually need to refresh it wherever instructor view used to remake the plugin, but I think it will cause little to no harm to do that.

fjonath1, 2017-03-07:
This issue is fixed on commit [e6f386ba9f7aa3224ee36adfac3f71f01b0b9a9b](https://github.com/cpaxton/costar_stack/commit/e6f386ba9f7aa3224ee36adfac3f71f01b0b9a9b)

closed 2017-03-07 by fjonath1