cpaxton/costar_stack #56: Robotiq on Kinetic

cpaxton, 2017-03-11:


ahundt, 2017-03-11:
https://github.com/cpaxton/costar_stack/issues/10

we made kinetic-devel and indigo-devel selectable here:
https://github.com/jhu-lcsr/robotiq/tree/kinetic-devel

ahundt, 2017-03-11:
setup script:
https://github.com/ahundt/robotics_setup/blob/master/costar.sh

cpaxton, 2017-03-14:
setup script should be in this repo

cpaxton, 2017-03-14:
and there are no changes in the robotiq package you linked me to

ahundt, 2017-03-20:
There are definitely changes, look at these:
https://github.com/jhu-lcsr/robotiq/tree/kinetic-devel-merge-jade
https://github.com/jhu-lcsr/robotiq/tree/kinetic-devel
https://github.com/jhu-lcsr/robotiq/tree/jade-devel

Also there weren't branches for each release before.
If kinetic-devel-merge-jade can be tested and works on the robot we should probably use that.

As for the setup script sure I can copy paste it in...

closed 2017-11-14 by cpaxton