cpaxton/costar_stack #57: Disable and skip gripper functions when there are no gripper found

fjonath1, 2017-03-11:
If there is no gripper service found, CostarArm will wait indefinitely for the gripper service.

This will cause instructor could not do anything and just say "NO CONNECTION" even though the driver is running properly.

ahundt, 2017-03-12:
Also when this is fixed it should let the system move but also still have a warning that there is no gripper connection. :-)

cpaxton, 2017-03-14:
So I disagree -- I'd rather have the system fail if it's configured to launch a gripper and can't find one. When the refactor PR goes in, it will have a yaml/rosparam configuration file. If you want to disable a gripper it should go in there.

ahundt, 2017-03-20:
Not detecting the gripper shouldn't prevent users from editing their files.

fjonath1, 2017-03-20:
I think it would be great if instructor status has a smarter status window that shows whether each big components in CoSTAR is up or not. Rather than just saying robot not connected even if it is not actually robot connection issue.

For example, we can have a status window that shows our current robot status (teach/servo/connected/not connected), gripper status (connected/not connected), perception status (connected/not connected), etc. Then we can set the color of the robot status to green if all components is connected and something like yellow of some of the components are not up, and make an annoying warning behavior every time we execute the tree if system is not all green.

ahundt, 2017-04-19:
@fjonath1 I like that idea. It would be cool if there were a simple python gui which could start up the whole system, deal with some of the flags, start and restart things.

Maybe one of these?

- http://www.clearpathrobotics.com/guides/ros/Creating%20RQT%20Dashboard.html
- http://wiki.ros.org/rqt_launch
- http://wiki.ros.org/node_manager_fkie